peng-zhihui/Dummy-Robot #27: 有想要复现的同学一起拼个团吗？

Khala-Yang, 2021-10-08:
CNC金属外壳部分，不知道3D打印的可不可以 >_<

S1aoXuan, 2021-10-09:
准备跟组里的兄弟的复现但是硬件是着实不会>.<

Hconk, 2021-10-09:
建了个群有兴趣复现的可以一起讨论下(纯技术交流及成果分享),  希望可以使用其他的减速电机作为替代品，在尽量降低成本的同时能获得相对较高的精度， 一些物料如果批量做报价应该可以低些。 538103629

captain-zhang, 2021-10-09:
准备复现或者自制的同学可以加我啊，各方面都懂一点，可以团一个出来。QQ 741243893

Khala-Yang, 2021-10-09:
这么贵的吗！... 实属无知者无畏(捂脸)

billhhh, 2021-10-10:
举手，一起啊

xiangxing98, 2021-10-10:
赞一个

Khala-Yang, 2021-10-10:
逛了一圈GIthu，发现了有个便宜一点的 [SmallRobotArm](https://github.com/SkyentificGit/SmallRobotArm)，都是3D打印件，也没有用很贵的电机，预估千元以内，吃吃土还是拿的下的。
稚晖君的这个机械臂Sooo Sexy！但要复刻一个就真得喝西北风了...
我永远相信稚晖君的审美，可我的钱包不相信我>_<！

chenwh14, 2021-10-10:
初步看了一下模型，发现有CSF-8-30，CSF-5-30两种Harmonic的谐波减速器，全新的话不会便宜……二手的话应该也难有靠谱的渠道啊

Hconk, 2021-10-10:
> 初步看了一下模型，发现有CSF-8-30，CSF-5-30两种Harmonic的谐波减速器，全新的话不会便宜……二手的话应该也难有靠谱的渠道啊

从模型里面数了下用的电机。

电机 | 个数
-- | --
PK513PA-H50S | 3
CSF-5-30-1U | 2
CSF-8-30-1U | 2
Nema17 Stepper Motor | 2
CRK543AP-H50 | 1

peng-zhihui, 2021-10-10:
谐波的成本很高，其实我是希望有同学有空修改一下模型改成行星减速方案（其实也不难），然后3D打印机身，精度虽说和我视频中比不了，但是比一般的舵机机械臂是要强很多的，而且后面代码可以通用

SinKy-Yan, 2021-10-10:
其实行星的话感觉可以试试DJI的那两个给比赛做的电机，[M3508](https://www.robomaster.com/zh-CN/products/components/general/M3508)和[M2006](https://www.robomaster.com/zh-CN/products/components/general/M2006) 这两个的回差应该在可接受的范围内，不知道有没有大佬去看看

GitHubHun, 2021-10-11:
我来看看大家在聊什么我看不懂的东西;

Fanjustfan, 2021-10-12:
想要复现+1 大佬们有群吗 交流比较方便

liubog2008, 2021-10-13:
https://join.slack.com/t/just-for-funco/shared_invite/zt-wysbftd3-iguJ4uYk2n9TgtX6fmkx0A 发个 slack 链接，欢迎大家加入

willshion, 2021-10-13:
求行星 复现 1k左右踢我

captain-zhang, 2021-10-15:
> 求行星 复现 1k左右踢我

1K 你还是歇着吧

MuNineyi, 2021-12-14:
> https://join.slack.com/t/just-for-funco/shared_invite/zt-wysbftd3-iguJ4uYk2n9TgtX6fmkx0A 发个 slack 链接，欢迎大家加入

可以给个 slack 的邀请链接么，原连接过期了

fumingxi, 2021-12-20:
我3D打印了一套对应的金属件，强度还可以，不过对PCB之类的不通（纯软件出身）。。有没有大佬篇硬件的一起搞一下。？

StarJi1991, 2022-02-21:
查了一下这个电机价格是认真的？

52chengcheng, 2022-06-28:
有没有群，邀一下