peng-zhihui/Dummy-Robot #68: DH参数好像不对？是不是我搞错了？D_ELBOW不应该是连杆的长度a吗？为啥是d

guguobao, 2021-11-04:
[image]

peng-zhihui, 2021-11-04:
你搞错了，D是连杆偏移，L才是长度，注意我模型里面是有角度Offset的，3轴的角度旋转了90度

guguobao, 2021-11-04:
> 你搞错了，D是连杆偏移，L才是长度，注意我模型里面是有角度Offset的，3轴的角度旋转了90度
[image]
[image]
D_ELBOW连杆偏移好像是沿着关节3的Z轴偏移的是吧？
但我看实物D_ELBOW更像是连杆的长度，沿着关节3的x轴偏移。总觉得和实物不太符合？我用的是标准DH建模思路，不知道思路对不对？望大佬指点

guguobao, 2021-11-04:
我猜你的正解并没有使用到DH参数中的d和a，因此你的程序在使用中是没问题的，

guguobao, 2021-11-04:
[image]
想请教下L1_BASE、L2_ARM、L3_ELBOW、L6_WRIST这四个向量的初始值缘由？

peng-zhihui, 2021-11-04:
当然用了啊，不用怎么正逆解，代码里都有

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: guguobao ***@***.***&gt;
发送时间: 2021年11月4日 13:54
收件人: peng-zhihui/Dummy-Robot ***@***.***&gt;
抄送: 稚晖 ***@***.***&gt;, Comment ***@***.***&gt;
主题: 回复：[peng-zhihui/Dummy-Robot] DH参数好像不对？是不是我搞错了？D_ELBOW不应该是连杆的长度a吗？为啥是d (Issue #68)

peng-zhihui, 2021-11-04:
你把机械臂三轴抬起90度机械臂竖直一条线就会明白了

发自我的iPhone

------------------ 原始邮件 ------------------
发件人: guguobao ***@***.***&gt;
发送时间: 2021年11月4日 13:54
收件人: peng-zhihui/Dummy-Robot ***@***.***&gt;
抄送: 稚晖 ***@***.***&gt;, Comment ***@***.***&gt;
主题: 回复：[peng-zhihui/Dummy-Robot] DH参数好像不对？是不是我搞错了？D_ELBOW不应该是连杆的长度a吗？为啥是d (Issue #68)

peng-zhihui, 2021-11-04:
> [image] 想请教下L1_BASE、L2_ARM、L3_ELBOW、L6_WRIST这四个向量的初始值缘由？

坐标系是右手坐标，图中右为X+，上为Z+；那个d/a只是临时变量，没有用到，后面实际直接用DH_matrix里面的变量了

guguobao, 2021-11-04:
好的，我刚刚猜到了L1_BASE、L2_ARM、L3_ELBOW、L6_WRIST的定义的，是不是这样的，比如L1_BASE= {armConfig.D_BASE, -armConfig.L_BASE, 0.0f}，L1_BASE即是把D_BASE看成在R[0]旋转后的坐标系x轴方向的值,  L_BASE看成R[0]旋转后的坐标系y轴方向的值 , 
如图：

guguobao, 2021-11-04:
[image]

guguobao, 2021-11-04:
望指导，十分感谢

peng-zhihui, 2021-11-04:
> 望指导，十分感谢

是的理解正确，简单讲L都是本身臂长，D是和臂长垂直方向的偏移

peng-zhihui, 2021-11-04:
另外初始角度偏移90°是当时为了和RoboDK里面的参数匹配

guguobao, 2021-11-04:
好的，谢谢了，我还想问问逆解，有什么好的方法理解 关节2有四组解，每组解得几何推导，有点难理解

guguobao, 2021-11-04:
[image]

peng-zhihui, 2021-11-04:
可以看一下林沛群教授的视频课程，讲得很好

发自我的iPhone

guguobao, 2021-11-05:
我又来麻烦你了，林沛群得机器人学我看完逆解部分了，它仅仅提到欧拉角zyz，没有提到已知zyz欧拉矩阵去反推角度。zyz欧拉矩阵（好像就是等同rotz(theta4)*roty(theta5)*rotz(theta6)（右乘，相对旋转），正好对上关节4坐标z、关节5坐标系y、关节6坐标z），逆解前面代码我看懂了，求关节1（左右手）、关节2（正反手）和关节3（关节2确定，关节3也就确定了）我几何推出来了，现在看不懂R36矩阵是如何一步一步反推关节5、4、6，我参照《机器人学导论-原书第四版》得p34页，书上有明确得公式去反推计算欧拉角，但发现和代码中得不是很匹配，我想问下，是要深刻理解欧拉角旋转矩阵形成，不一定按书上公式也可以？但我不知如何理解它，如何理解你的atan2去代入R36矩阵得值一步一步去计算，书上得公式是怎么来的，书也没有说？好抽象，求指点或者书籍推荐，

msnh2012, 2021-11-05:
关于姿态学，可以看看我这个repo. https://github.com/msnh2012/Msnhnet/tree/master/src/math

sampletea, 2021-11-12:
基座坐标系 和 i1-i6坐标系分别建在哪里？  我画到i5就没找不着了

guguobao, 2021-11-12:
> 基座坐标系 和 i1-i6坐标系分别建在哪里？ 我画到i5就没找不着了

在关节6末端，最后一个连杆的末端，标准DH建模的坐标系在连杆的末端

sampletea, 2021-11-12:
> > 基座坐标系 和 i1-i6坐标系分别建在哪里？ 我画到i5就没找不着了
> 
> 在关节6末端，最后一个连杆的末端，标准DH建模的坐标系在连杆的末端

大佬能 不能画一整个建好的图 出来 我看看我哪里错了

closed 2021-11-14 by guguobao

Gzlaughing, 2021-12-11:
> 我又来麻烦你了，林沛群得机器人学我看完逆解部分了，它仅仅提到欧拉角zyz，没有提到已知zyz欧拉矩阵去反推角度。zyz欧拉矩阵（好像就是等同rotz(theta4)*roty(theta5)*rotz(theta6)（右乘，相对旋转），正好对上关节4坐标z、关节5坐标系y、关节6坐标z），逆解前面代码我看懂了，求关节1（左右手）、关节2（正反手）和关节3（关节2确定，关节3也就确定了）我几何推出来 [...]

就是矩阵运算，稚辉君的正逆解是用了3x3的矩阵，机器人学导论是4x4的。可以去看MIT公开课的线性代数，理解一下矩阵的几何含义。

a361251388, 2022-04-02:
> [image]

这个问题解决了吗，我也推算D_ELBOW是a？ 沿z2方向看，x2和x3之间d为0；沿x3方向，z2和z3的距离为D_ELBOW，旋转90度打直也也不对呀

LZD0502, 2022-04-26:
您好 逆解法的代码您能解答下吗？

MatMultiply(R06, L6_WRIST, L0_wt, 3, 3, 1); 乘出来的L0_WRIST是啥意思，这一步没怎么懂。

MatMultiply(R10, P0_w, P1_w, 3, 3, 1);这个R10不就是A1的逆嘛，乘出来的P1_w又是啥

LZD0502, 2022-04-29:
> 可以看一下林沛群教授的视频课程，讲得很好 发自我的iPhone
> […](#)

您好 逆解法这部分代码 有没有详细点的解释啊

tongchuyan, 2022-07-27:
> > [image]
> 
> 这个问题解决了吗，我也推算D_ELBOW是a？ 沿z2方向看，x2和x3之间d为0；沿x3方向，z2和z3的距离为D_ELBOW，旋转90度打直也也不对呀

你好，问一下解决了吗？我也觉得应该是a

tongchuyan, 2022-07-27:
> > 你搞错了，D是连杆偏移，L才是长度，注意我模型里面是有角度Offset的，3轴的角度旋转了90度
> > [image]
> > [image]
> > D_ELBOW连杆偏移好像是沿着关节3的Z轴偏移的是吧？
> > 但我看实物D_ELBOW更像是连杆的长度，沿着关节3的x轴偏移。总觉得和实物不太符合？我用的是标准DH建模思路，不知道思路对不对？望大佬指点

我也这么觉得，问一下解决了吗？

qq302011, 2022-09-20:
哪位大佬能否解释下这个第五个关节怎么l和d都是0呢？另外我通过software里面的仿真软件改变第六个关节的角度，没有发现任何变化？第六个关节在哪呢？

adsl7878778, 2023-01-05:
个人感觉是坐标系建立不同的问题，不过用大佬的DH参数进行仿真，确实会出现上述问题，看起来与实物不符，自己进行DH参数建立，和大佬的与一些不同之处，但在进行仿真的时候，各关节位置和实物是一致的，具体什么原因我也不是很清楚，可能体现在代码里？还没有看。
[image]
用大佬的参数仿真出来是这样的
[image]
用自己的DH参数仿真出来是这样的
[image]
也不知道哪个对，继续往下学习再说吧，如果能有大佬解释一下就最好了

hhjhz, 2023-08-07:
> 个人感觉是坐标系建立不同的问题，不过用大佬的DH参数进行仿真，确实会出现上述问题，看起来与实物不符，自己进行DH参数建立，和大佬的与一些不同之处，但在进行仿真的时候，各关节位置和实物是一致的，具体什么原因我也不是很清楚，可能体现在代码里？还没有看。 [image] 用大佬的参数仿真出来是这样的 [image] 用自己的DH参数仿真出来是这样的 [image] 也不知道哪个对，继续往下学习再说吧，如果能有大佬解释一下就最好了

我也是这样，请问问题解决了吗？