peng-zhihui/Dummy-Robot #72: 这里有 几个问题 梯形加减速规划 是写在步进驱动板里的啊? 好像CAN仅发了目标角度和 时间啊？

sampletea, 2021-11-09:
dummy.UpdateJointPos();  这里边是获取每个电机旋转了多少°吗？ float time = dummy.MoveJ(pose[0], pose[1], pose[2],
                                     pose[3], pose[4], pose[5]); 这  6个pose   是指每个电机 的目标角度 ？ 
 怎么 执行MoveL的 时候 需要 进行一边 逆解dof6Solver->SolveIK(pose6D, lastJoint6D, ikSolves);  之后在跟着MoveJ  ;;           MoveJ 不需要调用SolveFK() 的么？     
 MoveJoints();    发送 目标角度 和 时间 到步进驱动器了啊  没看到梯形加减速规划呢  梯形加减速规划是写在 步进驱动板里的啊？

doche, 2021-11-13:
同问，没看到所说的梯形加减速，请指点，谢谢

peng-zhihui, 2021-11-17:
力矩、速度、位置、轨迹规划闭环都在驱动器里，因为我这里是总线控制，不是grbl那种脉冲同步控制，放在驱动器里才是合理的

sampletea, 2021-11-17:
> 力矩、速度、位置、轨迹规划闭环都在驱动器里，因为我这里是总线控制，不是grbl那种脉冲同步控制，放在驱动器里才是合理的

好的吧；  ！！   剩下的还能 解答一下吗？

vinsono, 2024-01-16:
SetPositionWithVelocityLimit(float _pos, float _vel)角度增量和最大速度到驱动，每个轴单独做插补和梯形加减速，没有做多轴联动的插补，所以无法实现直线和圆弧这种运动。

CoderSuHang, 2024-01-16:
您好，我是苏航，感谢您的来信！我会尽快对来信进行查看并尽快给您进行回复，祝您身体健康，工作顺利，谢谢！