peng-zhihui/Dummy-Robot #73: 3D模型文件打不开，使用什么软件？

ouyangxin95, 2021-11-17:
Dummy v164.step这个文件我用SolidWorks2016打不开。请问你们是用的什么软件设计，CAD吗？

Colin-Ji, 2021-11-17:
.step文件应该是３维文件的通用交换格式，一般的三维软件应该都可以，你看下你的导入方式，不行的话用其他的软件试下

主题: [peng-zhihui/Dummy-Robot] 3D模型文件打不开，使用什么软件？ (Issue #73)

echoeswang, 2021-11-18:
你随便 我用这个 Fusion 360

PancenHuang, 2021-11-19:
是不是你的SoildWorks软件版本比较旧，好像是比较旧的版本无法打开新版本画的图

AkrumIdrees, 2021-11-29:
I faced the same problem, the file is corrupt 
I used SolidWorks and Fusion on multiple computers with no Luck!

PancenHuang, 2021-11-29:
> I faced the same problem, the file is corrupt I used SolidWorks and Fusion on multiple computers with no Luck!

My dear friend，Which version did you choose.My SoildWorks is 2020,Try changing the version？

AkrumIdrees, 2021-11-29:
> > I faced the same problem, the file is corrupt I used SolidWorks and Fusion on multiple computers with no Luck!
> 
> My dear friend，Which version did you choose. My SoildWorks is 2020,Try changing the version？

I used SW 2017, SW2019 and Fusion latest Version. All Failed. I suggest uploading in Parasolid X_T., its is more stable than STEP.

PancenHuang, 2021-11-29:
> I used SW 2017, SW2019 and Fusion latest Version. All Failed. I suggest uploading in Parasolid X_T., its is more stable than STEP.
You can try a different version，Maybe  he used SoildWorks 2020 to draw the Model.In Our life we usually export STEP for easy.

Philips-Luo, 2022-04-06:
> I used SW 2017, SW2019 and Fusion latest Version. All Failed. I suggest uploading in Parasolid X_T., its is more stable than STEP. You can try a different version，Maybe he used SoildWorks 2020 to [...]

My dear friend, I can read files normally with PROE5.0 without any problem.

chedy028, 2022-04-11:
I use sharp3d, can not open the file as well.

fumingxi, 2022-04-12:
Try Fusion 360.

chedy028, 2022-04-13:
> 

Hey, Thanks. I tried, still not working

fumingxi, 2022-04-14:
Could you please attach the screenshot of error while opening?
I just downloaded the latest one and verified it was working.

Philips-Luo, 2022-04-14:
Dear friends,

[PENG-ZHIhui/dummy-Robot]: Dummyv164.step is designed using Fusion360 and exported as a. Step file. You can use other 3D model software and open it as an import. I use Pro/Engineer5.0 to open the file directly.

Philips-Luo, 2022-04-14:
Dear friends,

[PENG-ZHIhui/dummy-Robot]: Dummyv164.step is designed using Fusion360 and exported as a. Step file. You can use other 3D model software and open it as an import. I use Pro/Engineer5.0 to open the file directly.

抄送: "ph ***@***.******@***.***&gt;;

chedy028, 2022-04-14:
> Could you please attach the screenshot of error while opening?
> 
> I just downloaded the latest one and verified it was working.

[image]

Philips-Luo, 2022-04-14:
chedy028:

          Yes, but I can only take screenshots for you in my free time.

抄送: "ph ***@***.******@***.***&gt;;

chedy028, 2022-04-14:
[](url)> Could you please attach the screenshot of error while opening? I just downloaded the latest one and verified it was working.
[image]
Another screenshot. The file was corrupted. don't know why, tried both on Mac and Ipad. 

I probably should use Windows.

chedy028, 2022-04-14:
> > Could you please attach the screenshot of error while opening? I just downloaded the latest one and verified it was working. [image] Another screenshot. The file was corrupted. don't know why, tried both on Mac and Ipad.
> 
> I probably should use Windows.

I fixed it !!!!!. simply git clone rather than download it directly from the github.