peng-zhihui/Dummy-Robot #82: 请教一下.exe和dll也算开源吗，那是不是Windows也是开源操作系统了？

LordCasser, 2022-01-04:


peng-zhihui, 2022-01-04:
眼睛不用可以捐给有需要的人，“仓库里有dll”等于“仓库里只有dll”？Firmware里面完整代码看不到？仓库代码统计看不到？
C 94.3% C++ 3.8% ASP.NET 1.5% Other 0.4% ？
仓库README说明看不到？
[image]

peng-zhihui, 2022-01-04:
我也可以选择不上传exe，你去用license 3万一个的roboDK就是了，上传自己做的免费软件给你用你还白嫖出优越感了？代码核心都在控制器固件和驱动器固件里，我都完全开源了你视而不见？一个上位机就是个GUI壳子你搁这避重就轻是想带什么节奏？

peng-zhihui, 2022-01-04:
你还“trying to get better at Hardware”，所以仓库的完整Hardware源文件你看都不看一眼？眼里就只有那几个DLL了？除了找茬我想不出你提这个issue别的理由啊

LordCasser, 2022-01-04:
1. 其他事情做的好，不能掩盖上位机做法错误问题
2. 最初视频中声称演示所用上位机完全自己开发，将会开源，但是视频中的是roboDK
3. README中的《关于上位机》最初半点没提到，可以理解为补救措施
4. 对自己实现上位机的质疑，申称会开源，开源在何处？
5. 开源不是给狗或者乞丐丢骨头，爱开不开，没人逼着你开源，　开源　！＝　施舍
6. 至于白嫖，不好意思，我根本不用
7. ”因为有功能要添加所以添加好了才开源“，完全站不住脚，大可以开源出来大家一起完成，这才是开源的意义，难不成是代码有问题，比如注释没删干净？

peng-zhihui, 2022-01-04:
> 1. 其他事情做的好，不能掩盖上位机做法错误问题
> 2. 最初视频中声称演示所用上位机完全自己开发，将会开源，但是视频中的是roboDK
> 3. README中的《关于上位机》最初半点没提到，可以理解为补救措施
> 4. 对自己实现上位机的质疑，申称会开源，开源在何处？
> 5. 开源不是给狗或者乞丐丢骨头，爱开不开，没人逼着你开源，　开源　！＝　施舍
> 6. 至于白嫖，不好意思，我根本不用
> 7. ”因为有功能要添加所以添加好了才开源“，完全站不住脚，大可以开源出来大家一起完成，这才是开源的意义，难不成是代码有问题，比如注释没删干净？

我也没说上位机要开源啊，你编的？还是你想象的？功能添加完了我也不会开源

“README中的《关于上位机》最初半点没提到，可以理解为补救措施”，你是看不到Update日期？3个月前就上传并更新readme了，眼睛可以多用用。
[image]

你不用那就更说明你就是找茬的动机了啊，你不用有的是人用，很多人觉得受益不包括你，开源不是给狗的那你又不用还硬贴过来是想表达啥？

peng-zhihui, 2022-01-04:
另外所谓的“最初视频中声称演示所用上位机完全自己开发，将会开源，但是视频中的是roboDK”，也是你编的？
原话是“我也开发了自己的上位机”，要不要去再仔细看看视频？何况我买了RoboDK的license，基于它二次开发了driver，算不算开发？视频里有没有说“完全自己开发”几个字？

peng-zhihui, 2022-01-04:
> “开源不是给狗或者乞丐丢骨头，爱开不开，没人逼着你开源，　开源　！＝　施舍”

你搁这精神分裂呢？提这个issue就是在质疑为啥上位机不开源，到这里又说爱开不开没人逼，这逻辑我也是服。

你怎么不去质疑为啥RoboDK不开源啊，开源出来大家一起完善不是更好吗？还收几万块钱的license，说不定是因为里面也有代码注释没删干净吧对不对？

LordCasser, 2022-01-04:
遣词用句上我不太严谨，见谅
至于去搜寻证据来驳斥，或者玩文字游戏，我没有功夫，但是有很多有闲的人已经去做了，好像那种话题下你就不回复了？
视频剪辑的很精致，全程未露上位机软件标识，一边用一边说，”我也开发了类似于RoboDK的DummyStudio”，有人质疑后才说自己开发的未达标，视频演示用的RoboDK
文字游戏怎么玩都可以，但那是政客的把戏，不该是追求技术的人的手段

peng-zhihui, 2022-01-04:
> 遣词用句上我不太严谨，见谅 至于去搜寻证据来驳斥，或者玩文字游戏，我没有功夫，但是有很多有闲的人已经去做了，好像那种话题下你就不回复了？ 视频剪辑的很精致，全程未露上位机软件标识，一边用一边说，”我也开发了类似于RoboDK的DummyStudio”，有人质疑后才说自己开发的未达标，视频演示用的RoboDK 文字游戏怎么玩都可以，但那是政客的把戏，不该是追求技术的人的手段

是啊，做技术的关注技术就行了，这个项目95%的内容在结构设计，在控制算法，在硬件电路，上位机占比不到5%，视频里也就出现了十多秒钟，而你却只关注到这一个地方，到底是谁在带节奏你看不出来？你是搞技术的，有人是搞事情的，被当做枪使了而且你还信以为真。

抠字眼的是那些人，我上传视频分享作品一不带货二不卖课，还绝大部分完整开源，而有的人却只知道吹毛求疵找茬。硬要找茬的话你一句话超过10个字我就可以找出bug，比如“开源不是给狗或者乞丐丢骨头，爱开不开”，要是有人想搞事情是不是可以给你扣个帽子说你暗示其他用开源的人是狗？这有意思吗？

zhaihao118, 2022-01-04:
> 遣词用句上我不太严谨，见谅 至于去搜寻证据来驳斥，或者玩文字游戏，我没有功夫，但是有很多有闲的人已经去做了，好像那种话题下你就不回复了？ 视频剪辑的很精致，全程未露上位机软件标识，一边用一边说，”我也开发了类似于RoboDK的DummyStudio”，有人质疑后才说自己开发的未达标，视频演示用的RoboDK 文字游戏怎么玩都可以，但那是政客的把戏，不该是追求技术的人的手段

没必要这么较真，上位机只是一小部分，而且核心逻辑在固件中都有，视野放宽点。

peng-zhihui, 2022-01-04:
> 

不只是RoboDK，视频中Altium designer、Fusin360我都没有露出软件名称，什么是广告嫌疑你懂吗？何况花了钱买的软件我爱怎么用就怎么用有义务宣传吗？

peng-zhihui, 2022-01-04:
> > 遣词用句上我不太严谨，见谅 至于去搜寻证据来驳斥，或者玩文字游戏，我没有功夫，但是有很多有闲的人已经去做了，好像那种话题下你就不回复了？ 视频剪辑的很精致，全程未露上位机软件标识，一边用一边说，”我也开发了类似于RoboDK的DummyStudio”，有人质疑后才说自己开发的未达标，视频演示用的RoboDK 文字游戏怎么玩都可以，但那是政客的把戏，不该是追求技术的人的手段
> 
> 是啊，做技术的关注技术就行了，这个项目95%的内容在结构设计，在控制算法，在硬件电路，上位机占比不到5%，视频里也就出现了十多秒钟，而你却只关注到这一个地方，到底是谁在带节奏你看不出来？你是搞技术的，有人是搞事情的，被当做枪使了而且你还信以为真。
> 
> 抠字眼的是那些人，我上传视频分享作品一不带货二不卖课，还绝大部分完整开源，而有的人却只知道吹毛求疵找茬。硬要找茬的话你一句话超过10个字我就可以找出bug，比如你说的“开源不是给狗或者乞丐丢骨头，爱开不开”，要是有人想搞事情是不是可以给你扣个帽子说你暗示其他用开源的人是狗？这有意思吗？

peng-zhihui, 2022-01-04:
> > 
> 
> 不只是RoboDK，视频中Altium designer、Fusin360我都没有露出软件名称，什么是广告嫌疑你懂吗？何况花了钱买的软件我爱怎么用就怎么用有义务宣传吗？

质疑别人误导的同时自己却坚定地做着误导别人的事情，为什么你说的那些搜寻证据来驳斥的人发帖都要匿名+筛选评论你心里没点数？

peng-zhihui, 2022-01-04:
> > > 
> > 
> > 
> > 不只是RoboDK，视频中Altium designer、Fusin360我都没有露出软件名称，什么是广告嫌疑你懂吗？何况花了钱买的软件我爱怎么用就怎么用有义务宣传吗？
> 
> 质疑别人误导的同时自己却坚定地做着误导别人的事情，为什么你说的那些搜寻证据来驳斥的人发帖却自己匿名+筛选评论你心里没点数？

HSqure, 2022-01-04:
> 遣词用句上我不太严谨，见谅 至于去搜寻证据来驳斥，或者玩文字游戏，我没有功夫，但是有很多有闲的人已经去做了，好像那种话题下你就不回复了？ 视频剪辑的很精致，全程未露上位机软件标识，一边用一边说，”我也开发了类似于RoboDK的DummyStudio”，有人质疑后才说自己开发的未达标，视频演示用的RoboDK 文字游戏怎么玩都可以，但那是政客的把戏，不该是追求技术的人的手段

不会好好说话我来教教你吧，比如你说的“玩文字游戏”，但凡开始扣帽子了，就已经不在正常讨论的范畴里了，既然不想好好讨论，就别一本正经装讨论浪费大家时间了，这社区没你会更美好。

CaoZengxu, 2022-01-04:
> 最初标题“请教一下.exe和dll也算开源吗，那是不是Windows也是开源操作系统了？” 叙述有误，特此道歉 至于本项目的其他问题，本人对线经验浅薄，前期证据收集不足。无法挑战历经考验，对线经验极其丰富的“peng-zhihui”，也不想浪费时间去玩文字游戏。 我做不到的事情，欢迎大家一起完成，如果有人补充证据，欢迎在此发布，该ISSUE不会关闭
> 
> **peng-zhihui** ：我也不会关闭这个ISSUE，大家公正评价就行。**“预设一个立场和结论，然后再去找证据；能有理有据说事实=对线经验极其丰富”**，这大概是很符合你的开源精神和诚信理念的。 顺便我帮你把这个ISSUE置顶了。

你自己主张的自己不举证，全称凭空捏造张口就来。还这么理直气壮的让别人帮你找 牛的🐮

HSqure, 2022-01-04:
> 你自己主张的自己不举证，全称凭空捏造张口就来。还这么理直气壮的让别人帮你找 牛的🐮

笑死，经典卖惨“我弱我有理”：“嘤嘤嘤，我虽然没有证据，但是可以让大家帮弱小无力的我找找”（谁理他啊）

CRYOSTATx, 2022-01-04:
” 文字游戏怎么玩都可以，但那是政客的把戏，不该是追求技术的人的手段 “ ，那请问你又是在干什么？ 凭空捏造，张口就来是你所谓技术人的手段？ 可笑至极！

LordCasser, 2022-01-04:
https://www.zhihu.com/answer/2161285400

andy003, 2022-01-04:
> https://www.zhihu.com/answer/2161285400

所以，你想表达的是什么？
[image]
要不你资助点经费，把这个项目再打磨打磨？

andy003, 2022-01-04:
> https://www.zhihu.com/answer/2161285400 感谢置顶，诸多问题已经有人分析，期待一一批驳，至于文字游戏，图中显而易见 最初标题“请教一下.exe和dll也算开源吗，那是不是Windows也是开源操作系统了？” 叙述有误，特此道歉 至于本项目的其他问题，本人对线经验浅薄，前期证据收集不足。无法挑战历经考验，对线经验极其丰富的“peng-zhihui”，也不想浪费时间去玩文字游戏。 我做不到的事情，欢迎大家一起完成，如果有人补充证据，欢迎在此发布，该ISSUE不会关闭 [image]
> 
> **peng-zhihui** ：我也不会关闭这个ISSUE，大家公正评价就行。**“预设一个立场和结论，然后再去找证据；能有理有据说事实=对线经验极其丰富”**，这大概是很符合你的开源精神和诚信理念的。 顺便我帮你把这个ISSUE置顶了。

话说这不是个技术交流的地方？你觉得哪个地方，做的不好，可以自己做个，让大家膜拜膜拜。

peng-zhihui, 2022-01-04:
> https://www.zhihu.com/answer/2161285400

是的这就是我说的，“匿名+精心筛选评论的帖子”啊，黑子的钱赚得心安理得？你也就信信这样的言论了吧，放着众多理性评价的回答不看，选择性失明好不容易挑出一个抹黑的帖子拿来秀，需要靠各种动机揣测证明你的观点？

HSqure, 2022-01-04:
> https://www.zhihu.com/answer/2161285400
[image]

REFERENCES:
[1]. 弹幕
[2]. 弹幕
[3]. 网友评论
[4]. 网友评论

andy003, 2022-01-04:
> https://www.zhihu.com/answer/2161285400

[image]

@LordCasser 这位大佬，不用再纠结上位机是否开源了，你在这里找“证据”的时间，还不如花点时间学下C#或者QT，两周把这个上位机搞定？

peng-zhihui, 2022-01-04:
嗯欲加之罪要找茬很容易的。FOC视频里我一是提了有参考ODrive和VESC，但他们会再强调一遍这是参考开源项目的，说得好像你就是故意的一样；ODrivetool他们大概自己跑都没跑过，看着界面一样就说你是抄的，殊不知原版仓库是几年前发的现在连协议都不兼容了软件根本跑不起来，而我是参考它的Qt UI文件底层全部自己改掉才用起来的；所有项目开不开源，部分开源还是全部我都是明确说明的，FOC那个我有说过开源吗？仓库已经开源的FOC-Lite你倒是去看看啊，选择性失明和装睡的人我也是佩服的。

msnh2012, 2022-01-04:
人红是非多，或许一切都是开源惹的锅，如果一开始不说开源，就当节目看来也挺好.

peng-zhihui, 2022-01-04:
> 人红是非多，或许一切都是开源惹的锅，如果一开始不说开源，就当节目看来也挺好.

是啊，如果都是些这样的人也许还是不开源比较好

Jzacky, 2022-01-04:
> > 人红是非多，或许一切都是开源惹的锅，如果一开始不说开源，就当节目看来也挺好.
> 
> 是啊，如果都是些这样的人也许还是不开源比较好

不要啊，不要被这种人影响心情，跟着你做了很多项目，还想继续向您学习呢

LordCasser, 2022-01-04:
对对对，都是迫害，都是子虚乌有，当没摆出东西来的时候，攻击没摆出东西，摆出来了就是人红是非多
为什么不想想作为成名作的自行车项目没人说，其他项目被拔了一堆皮下来？

LordCasser, 2022-01-04:
> > https://www.zhihu.com/answer/2161285400
> 
> 是的这就是我说的，“匿名+精心筛选评论的帖子”啊，黑子的钱赚得心安理得？你也就信信这样的言论了吧，放着众多理性评价的回答不看，选择性失明好不容易挑出一个抹黑的帖子拿来秀，需要靠各种动机揣测证明你的观点？

那请一一驳斥，当驳斥不了的时候扣上帽子，打入另类，这种行为只能证明心虚和确有其事
抄开源不标注的时候申称自己是业余主播，难免没注意和疏漏，被扒皮时候就是黑你能赚钱，请问怎么通过黑一个业余主播赚钱？

peng-zhihui, 2022-01-04:
> 那请一一驳斥，当驳斥不了的时候扣上帽子，打入另类，这种行为只能证明心虚和确有其事 抄开源不标注的时候申称自己是业余主播，难免没注意和疏漏，被扒皮时候就是黑你能赚钱，请问怎么通过黑一个业余主播赚钱？

这不是有人驳斥了吗，你也看看兼听则明啊：https://www.zhihu.com/question/491080130/answer/2165261035   

我主业是AI工程师，不是业余难道是全职UP？
请问你这摆出来的不都是“动机揣测”吗？有理有据的话需要“匿名”+“精选评论”？所谓“一堆皮”除了你说的机械臂上位机（我自己实现了）和FOC（我实现并且开源了），还有啥？3年前做的plotclock？这东西我可就是写了个教程连视频都没做啊？另外成名作也不是自行车谢谢，量子计划小电视有人说了吗？HoloCubic？PocketLCD呢？仓库还有几十个开源项目呢你也没说啊。

Alansortorz, 2022-01-04:
不要被这些人影响心情，有啥事不能好好说，非要上来就夹带攻击的语调，如果不是动了某些人的蛋糕，那就是心态有问题，纯粹想杠。

Mazeqi, 2022-01-04:
呜呜呜 支持稚晖君

LongtsingLiao, 2022-01-05:
> > > https://www.zhihu.com/answer/2161285400
> > 
> > 
> > 是的这就是我说的，“匿名+精心筛选评论的帖子”啊，黑子的钱赚得心安理得？你也就信信这样的言论了吧，放着众多理性评价的回答不看，选择性失明好不容易挑出一个抹黑的帖子拿来秀，需要靠各种动机揣测证明你的观点？
> 
> 理性评论 = 维护你的评论，这是理性吗 帖子不是摆证据？有不理性的辱骂和扣帽子？ 那请一一驳斥，当驳斥不了的时候扣上帽子，打入另类，这种行为只能证明心虚和确有其事 抄开源不标注的时候申称自己是业余主播，难免没注意和疏漏，被扒皮时候就是黑你能赚钱，请问怎么通过黑一个业余主播赚钱？

稚晖对你关于开源的质疑都已经全部回复了，但你却到现在坚持给他扣个违背开源协议的帽子，而且在这个issue下有恶意言语攻击别人，而不是正常的技术交流，我想不通你不是引战的还能干啥，不要继续回复了好吗？浪费别人的时间，来github提issue的大多是想要交流技术的人，想引战别在这继续了，还我们一个干净的社区，好吗，别脏人了，太恶心了

iamike, 2022-01-05:
其实，看到你们在争论，我突然有个感觉：🤔
1. 🧬基因编辑工程……
2. 🤖工程机械军事装备……
3. 📞通讯设备数据库技术……
4. ……
💡结论：碳基生命(自私的基因) VS 硅基生命(彻底开源)
从哲学角度来看碳基生命为了基因的延续，必然会采取主动措施进行保护。其实，这也不是什么坏事，但是最终我们可能最终走向硅基生命吗？比如元宇宙😶‍🌫️🤯

mailonghua, 2022-01-06:
个人觉得，开源本身就是一个令人钦佩的精神，能从中学到很多知识，能够快速进步，这是一件非常有意义的事情，所以请不要吹毛求疵，鸡蛋里挑骨头的去刷存在感。请尊重别人的开源和努力。

iamike, 2022-01-06:
[image]
哈萨克斯坦又闹了，乌克兰和俄罗斯也闹矛盾，以前好好的一个社会主义大家庭👪，怎么就。。。。。变成今天这样

guguobao, 2022-01-12:
支持稚晖君

guguobao, 2022-01-12:
支持稚晖君 +1

cjx666666, 2022-01-18:
支持稚晖君 +2

NM2012, 2022-02-10:
> [https:](https://www.zhihu.com/answer/2161285400) //www.answer/28置顶问题，期待 5 个问题，一直在线，至于文字，最多 40 **人**[。 49180130/answer/2165261035](https://www.zhihu.com/question/491080130/answer/2165261035)有人帮忙批驳，兼听则明？
> 
> 最初的标题“请教一下.exe和也算开源吗，那不是Windows也是开源的吗？”叙述有误， 特此道歉至于本项目的其他问题，个人对线经验浅薄，接受不足。挑战历经，文字对线传奇 的“事件”，也不想浪费时间去玩游戏。会关闭 [image]
> 
> **peng-**：我也不会公允的举手，大家评价就行**了**。的开源精神和理念的。顺便帮 我顶一下这个诚信ISSUE。
> 
> [image]

又没要你看，不爽去别的地方。再说你不服你也整个出来给我们看看！

fanxing-6, 2022-03-14:
啊这, 怎么到处都有魔怔人, 没事找事....

845184278, 2022-03-16:
这不是技术人员逛的地方吗？这都是魔怔了

HeYongxian, 2022-04-27:
支持彭老师！

chi-xinming, 2022-06-23:
哇靠 这都有小黑子