peng-zhihui/Dummy-Robot #84: 驱动校准和REF通讯问题

learner-f, 2022-01-07:
关于电机校准部分

1. 烧录您直接提供的固件，按下KEY1 KEY2上电，电机进行正反方向运动，过程需长按按键否则运动终止，结束后，电机轴和未上电时是同样状态，重新上电重复上述操作，电机不再运动

2. 通过CLION直接编译烧录，短按KEY1 KEY2上电，电机就可以完成运动，结束后，可以外力转动电机出轴但能明显感受到阻力，且断电后重复按KEY1 KEY2上电，电机可再次运动

想问一下哪个是正常情况，若两者都有问题，希望大佬能指明应该是怎样的效果

与REF连接后，通过命令行发送...joint_1.do_calibration()命令，电机没有反应
[image]
芯片序列号已经通过修改程序，调试电机ID为1，除此之外，程序还有需要结合硬件调整的地方吗
程序逻辑梳理过了，细处不知从何入手，希望大佬能提供个解决思路，感谢

peng-zhihui, 2022-01-07:
1.应该都是正常的，电机校准完会自动重启，而上电不给指令的话是不会进入使能状态的（除非设置了enable_on_boot），所以你要通过CAN发送任意指令（位置、速度、力矩）才能让电机使能。
2.检查CAN是否连接正确（CAN1不是CAN2），终端电阻加了没（120R）

peng-zhihui, 2022-01-07:
另外不要用ODriveTool连接REF，协议不通用的，我新上传一个CMD根据你再试试

peng-zhihui, 2022-01-07:
电机你可以修改一下代码，在main.cpp里面用代码给一个速度指令来测试，具体指令函数可以参考can_protocol里面的内容

learner-f, 2022-01-14:
大佬你好，还有些问题请教

1. 最开始对源码直接编译时出了些错误，是mt6816_base.h和tb67h450_base.h文件中“端口指定工具”部分下的函数出现了“功能未实现”的错误
于是我在两个文件同名的CPP文件中重新添加这些函数，程序能正常编译了，但不知道这种方式会不会导致某些功能出错

2. ODrivetool中可以读取到REF的一些参数信息比如ID、温度等，所以“协议不通用”指的是对电机指令控制部分吗

3. 驱动校准后，按键1、2不能执行相应功能，这种情况貌似可以用issue#85中的内容解释，具体原因就不清楚了

关于硬件的部分已经检查过了，没有遗漏的地方
问题有些琐碎，见谅

peng-zhihui, 2022-01-17:
1. 把未实现的函数设置为纯虚函数就行（加上"=0"），应该是你的编译器问题。
2. 是的，电机协议是自定义的，跟odrivetool用的都是fibre实现所以底层基础协议通用但是上层应用协议不一样
3. 程序覆盖的问题我在issue#85回答了你再试一下，如果编译出的固件尺寸没问题应该就可以用了，或者你用我预编译的先测试一下（记得CAN要加终端电阻）

jzijin, 2022-04-13:
> 

稚晖君，你好。我当前也是REF连接电机驱动板调试。CAN通信失败，电机驱动板没有收到消息。
关于终端电阻，我看原理图上是都没有加终端电阻的。不加终端电阻是否会影响到CAN通信呢？

jzijin, 2022-04-13:
> 关于电机校准部分
>
> 1. 烧录您直接提供的固件，按下KEY1 KEY2上电，电机进行正反方向运动，过程需长按按键否则运动终止，结束后，电机轴和未上电时是同样状态，重新上电重复上述操作，电机不再运动
> 2. 通过CLION直接编译烧录，短按KEY1 KEY2上电，电机就可以完成运动，结束后，可以外力转动电机出轴但能明显感受到阻力，且断电后重复按KEY1 KEY2上电，电机可再次运动 [...]

你好。请问下后面这个问题解决了吗？是怎么解决的？望解答，感谢。

Joelliamliu, 2022-08-08:
> 

我的Dummy Studio跟核心板通讯一直有问题，显示能链接，有send信息但一直无receive信息，需要我下载Odrive安装吗？ 怎么才会出现Reconnected to Dummy-Robot 3055385E3030 as dummy0这条信息（我的显示
[image]
NameError: name 'dummy' is not defined）？谢谢

peng-zhihui, 2022-08-08:
用zadig安装libusb驱动，ref interface2那个设备替换为libusb驱动

Joelliamliu, 2022-08-09:
> 用zadig安装libusb驱动，ref interface2那个设备替换为libusb驱动

已经安装libusb驱动，但Dummy Studio和REF主板通讯依然有问题
[image]
[image]
[image]

peng-zhihui, 2022-08-09:
这个usb是reftool的命令行连接的，不是dummystudio，studio用的就是普通串口

Joelliamliu, 2022-08-09:
> 用zadig安装libusb驱动，ref interface2那个设备替换为libusb驱动

[image]

spacomp, 2022-08-18:
> Dummy Studio可以和REF通信了吗？

Joelliamliu, 2022-08-20:
> > Dummy Studio可以和REF通信了吗？

已经连接上，但通讯有问题，我正在找原因

ZXYZZP, 2022-08-26:
> 

请问找到原因了吗我的显示usb已连接但没接收到数据，发送数据也没反应

ZXYZZP, 2022-08-26:
> Joelliamliu

请问找到原因了吗我的也是usb连接成功但没接收到数据，发送数据也没反应