peng-zhihui/Dummy-Robot #85: 步进电机驱动器build出来的固件有119KB，无法再保存32K的编码器校验数据！

real-jianhua, 2022-01-12:
现在的代码build出来的固件有119K那么大，已经没有空间保存校验数据了，导致编码器校准后执行保存时把自身的固件给破坏了，程序重启后就没法正常工作了。

learner-f, 2022-01-13:
是电机校准之后就没有其他反应了吗，包括按键什么的

real-jianhua, 2022-01-13:
是的，固件被破坏了，无法正常工作了。

peng-zhihui, 2022-01-13:
我上一个提交应该就是修复这个问题了，你pull一下最新代码看看

real-jianhua, 2022-01-13:
我就是用最新提交的代码编译的了，目前这个代码编译出来的固件太大了，大约118K，我用烧录软件检查过烧录到stm32后，占用了0x8000000到0x801D8D0的空间，而编码器写入数据的起始地址是0x08017C00。这样保存编码器数据的话会把原来的固件破坏掉，保存重启后驱动器就没法正常工作了。

NM2012, 2022-01-15:
问下这个这个步进电机的代码是什么平台编译的？

peng-zhihui, 2022-01-16:
> 现在的代码build出来的.bin文件固件有119K那么大，已经没有空间再保存编码器的校验数据了，导致编码器校准后执行保存时把自身的固件给破坏了，程序重启后就没法正常工作了。

清除CMake缓存重新编译一下，我这边修复后是可以的：
[image]

StudenLin, 2022-03-06:
> 现在的代码build出来的.bin文件固件有119K那么大，已经没有空间再保存编码器的校验数据了，导致编码器校准后执行保存时把自身的固件给破坏了，程序重启后就没法正常工作了。

请问你的电机现在可以正常工作吗？如何查看固件是否被破坏了

Jzacky, 2022-03-23:
> 我就是用最新提交的代码编译的了，目前这个代码编译出来的固件太大了，大约118K，我用烧录软件检查过烧录到stm32后，占用了0x8000000到0x801D8D0的空间，而编码器写入数据的起始地址是0x08017C00。这样保存编码器数据的话会把原来的固件破坏掉，保存重启后驱动器就没法正常工作了。

那几个虚函数后边不要添加=0，否则换了编译器9.2.1编译出来也119，只保证编译器版本程序默认编译就可以了

ZXYZZP, 2022-08-12:
> 

我现在也遇到了相同的问题一定要更换编译器版本才有用吗