peng-zhihui/Dummy-Robot #89: 重力补偿问题

w-1016-Papman, 2022-01-15:
博主大大您好，看到您做的小型机械臂，非常好奇您是如何解决机械臂的重力补偿问题的，因为我想的是：在实际应用过程中，除开机械臂自身的重力力矩需要补偿，还有一些非几何误差是不易测量并且不易线性化的（比如连杆的形变、关节的柔性以及电机本身的精度阈值等等因素导致的末端执行器与l理论位置出现偏差），我已经想这个问题大概想了两周了，但很遗憾，我有点笨，所以来这里向您提问，希望您在看到这一问题时，能劳神帮我解答一二，感激不尽。预祝新年快乐！           

                                                                                                                                             ——   一名来自天津的大二本科生

peng-zhihui, 2022-01-17:
实际的机械臂零力拖拽主要就是消除两项：重力和摩擦力，因为拖动速度比较慢所以惯性力可以忽略；摩擦力可以认为只跟每个关节转速有关，重力要复杂一些，靠重力力矩回归矩阵描述，相关参数都需要做系统辨识通过最小二乘法标定出来，可以自己去搜一下相关资料和论文看。我这个项目里因为自重很轻，重力项是通过选取合适的减速器减速比靠静摩擦力矩平衡的。

msnh2012, 2022-01-19:
其实你说的末端和实际理论的误差，其实除了收到重力的影响之外，本身机械误差也是无法忽视的。一般来说，每台机械臂的DH参数都是不一样的，在出厂之前都需要使用激光跟踪仪来重新拟合DH参数。重力项对Dummy的影响不是很大，本身其负载也比较小，当然对于工业大负载的臂的话，重力，库伦摩擦力，惯性矩是需要通过参数辨识的，一般每个厂商都有自己一套标准的方法。当然做这些的前提的话，是需要一个不错的电机和减速器的，硬件不行的话，算法硬刚也是不行的。对于电机的话一般是PMSM搭配FOC实现精确力矩控制，对于减速器的话一般是谐波或者RV减速器达到小体积，"0"背隙，大减速比，大负载。

w-1016-Papman, 2022-01-19:
> 实际的机械臂零力拖拽主要就是消除两项：重力和摩擦力，因为拖动速度比较慢所以惯性力可以忽略；摩擦力可以认为只跟每个关节转速有关，重力要复杂一些，靠重力力矩回归矩阵描述，相关参数都需要做系统辨识通过最小二乘法标定出来，可以自己去搜一下相关资料和论文看。我这个项目里因为自重很轻，重力项是通过选取合适的减速器减速比靠静摩擦力矩平衡的。

好的，了解了，谢谢您嘞

w-1016-Papman, 2022-01-19:
> 其实你说的末端和实际理论的误差，其实除了收到重力的影响之外，本身机械误差也是无法忽视的。一般来说，每台机械臂的DH参数都是不一样的，在出厂之前都需要使用激光跟踪仪来重新拟合DH参数。重力项对Dummy的影响不是很大，本身其负载也比较小，当然对于工业大负载的臂的话，重力，库伦摩擦力，惯性矩是需要通过参数辨识的，一般每个厂商都有自己一套标准的方法。当然做这些的前提的话，是需要一个不错的电机和减速器的，硬件不行的话，算法硬刚也是不行的。对于电机的话一般是PMSM搭配FOC实现精确力矩控制，对于减速器的话一般是谐波或者RV减速器达到小体积，"0"背隙，大减速比，大负载。

好的好的，谢谢您啦！