peng-zhihui/Dummy-Robot #91: TRJ轨迹跟踪控制算法做了吗？

m986883511, 2022-01-19:
您在Readme中写支持TRJ轨迹跟踪
，我想参考您的实现，我找到比较像的是如下代码？是空的啊，是我找错了吗？
麻烦指导一下。

void DummyRobot::MoveTrajectoryJ(float _j1, float _j2, float _j3, float _j4, float _j5, float _j6)
{

}

void DummyRobot::MoveTrajectoryL(float _x, float _y, float _z, float _a, float _b, float _c)
{

}

peng-zhihui, 2022-01-19:
轨迹插值有一部分实现在驱动器内，控制器里面这个轨迹只做线性插值用于在两个MoveL指令间插值扩充指令频率到200Hz，另外这块我写完了还没时间测试，验证完成了会更新上来。

m986883511, 2022-01-19:
期待您的更新

m986883511, 2022-01-21:
在驱动器中的实现的线性插值代码，能指明一下在哪里吗？

还有一点建议：由于目前没有实现TRJ轨迹控制，建议您在Readme中标注位为 Todo。

peng-zhihui, 2022-01-21:
轨迹规划都在 `2.Firmware\Ctrl-Step-Driver-STM32F1-fw\Ctrl\Motor\motion_planner.cpp`文件里；意见已采纳
[image]