peng-zhihui/Dummy-Robot #95: 关于tb67h450_base.h

wztted, 2022-01-23:
tb67h450_base.h文件里好几个private函数都没有实现是怎么回事？？？

peng-zhihui, 2022-01-23:
复习一下c++基础，这是基类虚函数。

wztted, 2022-01-23:
> 复习一下c++基础，这是基类虚函数。

非常感谢大佬这么快回复（受宠若惊），是不是纯虚加上=0才可以？不然总报错。我是机械专业，在汽车行业已浸淫十年，这两年一直业余时间学习嵌入式方面的东西（单片机，PCB设计，C++），前年用stm32单片机做了一个魔方恢复机器人（抖音1704469178）。去年看到大佬的机械臂视频崇拜得五体投地，把你所有的视频都撸了一遍，准备亲手复现这个机械臂，希望实在搞不懂的时候能求助大佬，毕竟大佬日理万机。

zhenghengju, 2022-03-15:
> > 复习一下c++基础，这是基类虚函数。
> 
> 非常感谢大佬这么快回复（受宠若惊），是不是纯虚加上=0才可以？不然总报错。我是机械专业，在汽车行业已浸淫十年，这两年一直业余时间学习嵌入式方面的东西（单片机，PCB设计，C++），前年用stm32单片机做了一个魔方恢复机器人（抖音1704469178）。去年看到大佬的机械臂视频崇拜得五体投地，把你所有的视频都撸了一遍，准备亲手复现这个机械臂，希望实在搞不懂的时候能求助大佬，毕竟大佬日理万机。
我也是编译时 这几个函数报未定义  你最后是怎么解决的

wztted, 2022-03-15:
声明加上=0啊

zhenghengju, 2022-03-15:
> 
😅  对了还有这个工程怎么没有.ioc 文件

wztted, 2022-03-15:
> > 
> 
> 😅 对了还有这个工程怎么没有.ioc 文件

工程本来也不需要IOC啊，你要它干嘛？

zhenghengju, 2022-03-15:
> 
难道稚晖大佬本来就没用cubemx吗
看代码应该是用了cubemx 生成，定时器 dma 等等初始化，虽然看代码也能看出怎么配置的，有ioc文件不是方便看吗，而且也更方便修改。主要是对我这样的小白更友好  是不要求太高了😁