peng-zhihui/Dummy-Robot #99: 关于ref 最新固件和reftool工具的使用求助

StudenLin, 2022-02-12:
稚晖君您好，我尝试了前几天发布的ref v2.0编译通过，烧录后屏幕不亮，但接入usb有反应，电脑识别为ref 1.0 cdc interface，尝试使用reftool发现连接不上，折腾了一天还是不懂怎么弄，求救🆘大佬

[image][image]

peng-zhihui, 2022-02-14:
更新前可以连接吗？有没有自己添加额外的RTOS任务？有的话检查一下是否是栈溢出了

peng-zhihui, 2022-02-14:
需要用zadig安装libusb驱动的，你安装过了？

StudenLin, 2022-02-14:
> 需要用zadig安装libusb驱动的，你安装过了？

[image]

不知道这样子对不对呢

peng-zhihui, 2022-02-18:
对的，那你可能得自己debug一下了

StudenLin, 2022-02-18:
> 对的，那你可能得自己debug一下了

后面选interface 2就可以了，错选了interface 0，小错误浪费了好多时间😂，谢谢您

jzijin, 2022-04-12:
你好，请问下，我在执行run.bat的时候遇到这个问题，有可能是什么原因导致的呢？
[image]

Joelliamliu, 2022-08-15:
> 对的，那你可能得自己debug一下了
这个ACK：503代表什么信息？
[image]

ZXYZZP, 2022-08-17:
[image]
一连接板子就退出进程是什么原因

wastom, 2024-11-24:
> > 对的，那你可能得自己debug一下了
> 
> 后面选interface 2就可以了，错选了interface 0，小错误浪费了好多时间😂，谢谢您

[image]
[image]
我选了interface 2还是会断开连接，请问你有遇到过吗

CoderSuHang, 2024-11-24:
您好，我是苏航，感谢您的来信！我会尽快对来信进行查看并尽快给您进行回复，祝您身体健康，工作顺利，谢谢！