peng-zhihui/Dummy-Robot #109: 关于步进电机驱动代码中数字的疑惑

mailonghua, 2022-03-15:
作者你好，最近在看步进电机的驱动代码，但是在类似表示FOC控制的函数中存在疑惑若有空希望能帮忙解惑
1.如下面的将dac_reg*5083,这个5083该如何理解
2.同时想咨询一下，步进电机的软件细分有类似的文章能推荐一下吗，原理不了解，代码理解起来比较困难
非常感谢
_void TB67H450Base::SetFocCurrentVector(uint32_t _directionInCount, int32_t _current_mA)
{
    phaseB.sinMapPtr = (_directionInCount) & (0x000003FF);
    phaseA.sinMapPtr = (phaseB.sinMapPtr + (256)) & (0x000003FF);
    phaseA.sinMapData = sin_pi_m2[phaseA.sinMapPtr];
    phaseB.sinMapData = sin_pi_m2[phaseB.sinMapPtr];
    //dac_reg
    uint32_t dac_reg = abs(_current_mA);
    dac_reg = (uint32_t) (dac_reg * 5083) >> 12;
    dac_reg = dac_reg & (0x00000FFF);_

closed 2022-03-22 by mailonghua

jzijin, 2022-04-10:
请问下后来解决了吗？5083是什么意思呢？

raulcao, 2022-09-21:
> 请问下后来解决了吗？5083是什么意思呢？

5083 /4096 = 4095/3300

YangAoTeng, 2022-11-26:
(dac_reg * 4095 / 3300)的变种，这一段代码是根据细分来查sin表，决定PWM输出合成的电流，进而控制步进电机的运动。

YangAoTeng, 2022-11-26:
如果这一段实在看不懂的话，可以了解一下步进电机的结构与步进电机细分的原理。 就可以理解清楚了