peng-zhihui/Dummy-Robot #114: 电机驱动进入使能模式问题

Scoulson986, 2022-03-20:
稚晖君您好，在复刻42电机驱动板的时候我遇到了一点问题，期望您能答复：
驱动板现在可以进入编码器校准模式，电机正反转一圈后使能，阻力很大。但是key1和key2无反应，疑似死机现象。使用Stlink utility连接后阻力消失，key1和key2依然无反应。
检查驱动板焊接时测量了MT6816引脚的电压值，大致如下图所示。
校准完之后的数据似乎也没写入Flash中，在程序里看到MT6816的地址是0x08017C00，但当电机进入上述使能后，用utility看该地址的flash是空的。
后来试图烧录您的hex程序进驱动板，地址是有数据的，但是key1和key2摁下后仍然无反应。

图1（mt6816及其电压值）
[image]
图二（mt6816地址）
[image]
[image]

peng-zhihui, 2022-03-21:
用我的hex校准过后，不发送指令是不会转的，可以这样测试，按键1按一下解除使能，手动转动电机到任意位置，再按一下按键1使能，电机应该会回到初始位置

peng-zhihui, 2022-03-21:
你自己编译校准失败，应该是flash溢出的问题，翻一下之前的issues，解决办法是改用和我一样版本的GCC编译

Scoulson986, 2022-03-21:
好的，谢谢大佬，我再试试看

jzijin, 2022-04-09:
> 稚晖君您好，在复刻42电机驱动板的时候我遇到了一点问题，期望您能答复： 驱动板现在可以进入编码器校准模式，电机正反转一圈后使能，阻力很大。但是key1和key2无反应，疑似死机现象。使用Stlink utility连接后阻力消失，key1和key2依然无反应。 检查驱动板焊接时测量了MT6816引脚的电压值，大致如下图所示。 [...]

请问下，这个问题后来解决了吗？是怎么解决的？

Scoulson986, 2022-04-09:
这个问题最后解决了。说一下解决的方法，就是烧录完成之后使用debug模式，一步步运行，直到进入if条件(if button1和button2 pressed的条件）后，摁住key1，key2，继续运行，然后电机开始校正，继续单步运行直到完成一遍程序后，断电重启即可进行使能切换了。研究这个过程花了好长时间，最后推测可能是编译器版本原因或者环境配置和稚晖君的稍有出入，导致出现这个问题。

closed 2022-04-09 by Scoulson986