peng-zhihui/Dummy-Robot #121: 断电重启，电机驱动板无法正常运行

jzijin, 2022-04-08:
Hi，稚晖君：
请问下，有遇到过电机驱动板在烧录器烧录完之后可以正常运行，但是只要断电重启之后就无法正常运行的情况吗？

jzijin, 2022-04-08:
请问下电机驱动代码中，EXTI9_5_IRQHandler中断函数，为什么需要return之后在执行HAL_GPIO_EXTI_IRQHandler操作呢？
```
void EXTI9_5_IRQHandler(void)
{
  /* USER CODE BEGIN EXTI9_5_IRQn 0 */
    return;
  /* USER CODE END EXTI9_5_IRQn 0 */
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_7);
  /* USER CODE BEGIN EXTI9_5_IRQn 1 */

  /* USER CODE END EXTI9_5_IRQn 1 */
}
```

peng-zhihui, 2022-04-08:
应该是因为你编译的文件尺寸超过了flash设置的app区，导致校准之后数据覆盖了代码，参考issues里面另一个类似问题，替换gcc版本解决

jzijin, 2022-04-09:
> 应该是因为你编译的文件尺寸超过了flash设置的app区，导致校准之后数据覆盖了代码，参考issues里面另一个类似问题，替换gcc版本解决

稚晖君，你好。我debug的时候，发现掉电重启之后，频繁的走到了EXTI9_5_IRQHandler函数中，所以系统卡死了。我将return和HAL_GPIO_EXTI_IRQHandler交换顺序就好了。不知道这是为啥。

peng-zhihui, 2022-04-09:
那是IO信号的中断，这板子上没用到，直接删掉就行

closed 2022-04-10 by jzijin