peng-zhihui/Dummy-Robot #147: 电机ID参数问题

Joelliamliu, 2022-08-02:
稚晖君您好，修改CMake环境变量后对核心板REF用Clion进行了重新编译，但核心板与主板之间通讯依然有问题，是否需要修改电机ID参数？另外电机的序列号431466563640等是驱动板芯片的序列号还是REF主板J1-J6的跳线号？我刷驱动板程序的时候是否需要修改？还是自动是别的？谢谢！
[image]
[image]
[image]

peng-zhihui, 2022-08-02:
序列号和芯片硬件相关，需要自己先读取出来再修改到代码；也可以不管那个序列号，手动一个个电机设置地址就行了，默认都是0地址

Joelliamliu, 2022-08-02:
> 

明白了，非常感谢稚晖君不吝赐教！！！

Joelliamliu, 2022-08-05:
> 序列号和芯片硬件相关，需要自己先读取出来再修改到代码；也可以不管那个序列号，手动一个个电机设置地址就行了，默认都是0地址

[image]

yuZhangNathan, 2022-11-06:
> 

请问一下如何获取芯片地址

Zhongliang-liu, 2022-11-07:
下载一个TP-LINK，可以参考https://blog.csdn.net/Zhichao_Zhang/article/details/124411550