peng-zhihui/Dummy-Robot #150: 电机校准后死机问题

ZXYZZP, 2022-08-16:
hex文件下载进去后，按key1,key2上电，电机开始校准正转两圈反转一圈后停止，停止后按按键没办法使能或者使能电机向是死机了，重新按住按键上电毫无反应无法再进入校准状态.连上stlink utility查看flash发现原本用来储存编码器数据的0x08017C00地址被清空，我想应该是校准失败了但不清楚是什么原因。测试了一下自己编译的固件情况类似不同的是可反复进入校准状态。

ZXYZZP, 2022-08-16:
[image]

ZXYZZP, 2022-08-17:
[image]
调试发现程序由于不明错误进入了死循环导致卡死，现在还找不到原因

song-xuyu, 2023-01-09:
你好，请问大佬问题解决了吗？我也出现了这样的问题，就是只要一校准完成(貌似是校准失败)就会进入HardFault_Handler。

KeithBlackmore, 2023-01-10:
可以检查一下编码器使用的磁铁是不是径向充磁的，我更换磁铁后就可以了

Kevin75311, 2023-02-23:
> 
请问这个的意思是要在电机底部安装磁铁吗？

song-xuyu, 2023-02-23:
我用clion编译之后flash使用了98%，推测是不是可能cpp编译之后内存太大，校准之后数据存储擦除时将一些段破坏导致进入hardfault，使用xdrive的c版本没有问题

KeithBlackmore, 2023-03-02:
> > 
> 
> 请问这个的意思是要在电机底部安装磁铁吗？

是的，注意磁铁的物理特性，要径向充磁

Lan-Muzi, 2023-09-15:
> hex文件下载进去后，按key1,key2上电，电机开始校准正转两圈反转一圈后停止，停止后按按键没办法使能或者使能电机向是死机了，重新按住按键上电毫无反应无法再进入校准状态.连上stlink utility查看flash发现原本用来储存编码器数据的0x08017C00地址被清空，我想应该是校准失败了但不清楚是什么原因。测试了一下自己编译的固件情况类似不同的是可反复进入校准状态。

你好，请问这个问题你们如何解决的，我也遇到了这个问题，第一次电机校准的时候是正常的，key1使能/失能闭环都正常，但是后面自己编译了一个固件，改了电机ID，之后就出现这个问题了，能校准，但是校准完之后就不正常了，转动电机有阻尼感，不是完全转不动，但是key1按键完全不起作用，重新上电也是一样 不起作用。同时按住两个按键又可以进行校准，然后就是同样的现象了，换了稚晖君提供的固件也是一样的效果。

Lan-Muzi, 2023-09-15:
> 你好，请问这个问题你们如何解决的，我也遇到了这个问题，第一次电机校准的时候是正常的，key1使能/失能闭环都正常，但是后面自己编译了一个固件，改了电机ID，之后就出现这个问题了，能校准，但是校准完之后就不正常了，转动电机有阻尼感，不是完全转不动，但是key1按键完全不起作用，重新上电也是一样 不起作用。同时按住两个按键又可以进行校准，然后就是同样的现象了，换了稚晖君提供的固件也是一样的效果。

问题已经解决，原因是掉电重启之后进入错误中断引起，解决方式为在gpio.c文件中注释或者删除相关方法。
[image]