peng-zhihui/Dummy-Robot #155: 使用串口发送 p 命令 后，电机停转以后不停抖动大概20°角左右

bigbigbigpick, 2022-10-19:
使用串口发送 p 命令 后，电机停转以后不停抖动大概20°角左右，发送 V 命令 参数为零是可正常停下。 请问下是否是缺省参数问题。如何调整，谢谢。
[image]

bigbigbigpick, 2022-10-21:
> 确认硬件没问题的话，可能是参数整定的问题，需要自己多尝试几组数据调试下。默认参数是针对42步进电机的。

我是用的电机也是42步进电机，我在这个程序加了这个条语句，它就正常的运行了，造成这个现象可能是什么原因，还有晶振是12MHz 的吗？
[image]

peng-zhihui, 2022-10-21:
晶振是12MHz。那可能是编译器优化问题，你把release改成debug模式编译看看。

bigbigbigpick, 2022-10-21:
> 晶振是12MHz。那可能是编译器优化问题，你把release改成debug模式编译看看。

遇到了新问题，按照你的要求改成debug 编译时候出现了这种错误 undefined reference to`TB67H450Base::InitGpio()'。
 添加 =0 后 编译出来的代码太大了。达到了121K之多，没办法存放其他数据。会造成单片机死机，在release 模式下没有这个问题
[image]
编译器版本 应该和你的版本一样吧。能不能解答一下这是什么问题。

peng-zhihui, 2022-10-21:
flash最后要预留32K做校准数据储存区，debug好像是会超过尺寸的，你可以还是换回release模式，然后单独在calcSoftGoal函数前面加上`__attribute__((optimize("O1")))`字段。还是不行的话就需要你自己debug断点调试了，我远程看不出来啥问题。

zucc-zhaob, 2023-01-25:
> > 晶振是12MHz。那可能是编译器优化问题，你把release改成debug模式编译看看。
> 
> 遇到了新问题，按照你的要求改成debug 编译时候出现了这种错误 undefined reference to`TB67H450Base::InitGpio()'。 添加 =0 后 编译出来的代码太大了。达到了121K之多，没办法存放其他数据。会造成单片机死机，在release 模式下没有这个问题 [image] 编译器版本 应该和你的版本一样吧。能不能解答一下这是什么问题。

问题解决了吗？我也遇到这个问题。