peng-zhihui/Dummy-Robot #161: STM32F4 openocd开发，运行正常，但是debug过不去systemclock-config()函数

sytnocui, 2022-11-30:
稚晖君你好，我最近使用stm32f405rgt6开发项目，使用stlink，clion，openocd调试。发现debug模式无法通过systemclock-config函数，会卡在锁相环初始化这里。只有参考这个网址修改了系统生成的systemclock-config函数才可以正常调试。但是修改系统生成的函数，每次cubemx更新都会覆盖掉，非常不方便，不知道你是怎么解决这个问题的？
我的同学使用stm32cubeide调试，就没有这个问题。
我注意到你在瀚文键盘里的调试器已经改用了cmsis-dap，有没有这方面的考虑？

sytnocui, 2022-11-30:
[image]
如图所示，会跳进这里，然后卡死

peng-zhihui, 2022-12-01:
main函数最开始加一句RCC_DeInit()就行了，跟DAP没关系，时钟初始化的问题。

sytnocui, 2022-12-01:
> main函数最开始加一句RCC_DeInit()就行了，跟DAP没关系，时钟初始化的问题。

果然可以了，谢谢大佬orz。
另外我也尝试了在stm32cubeide里进行调试，这个调试方式是stlink gdb server，就是什么都不用加就是好使的。我尝试配了一个 clion+gdb+jlink 的debug环境，也是什么都不加就好使。
总结一下就是跟GDB沾边的都是能用的，但用openocd就得加“RCC_DeInit()”这句话。可否再麻烦问一下出现这种情况的详细原因？是和openocd有关系吗？

tip：如果有路人看到了可以参考这篇文章：[CLion配置STM32开发环境（JLink GDB Server）](https://blog.csdn.net/zjh1229/article/details/123887084 )

cmsjscj, 2024-07-07:
感谢万分！！另外我将诸多可能出现的意外情况进行汇总，路人有缘可以去参考下拙作[关于如何使用CLion对于STM32进行调试debug工作的讨论](https://blog.csdn.net/weixin_60664036/article/details/140252907?csdn_share_tail=%7B%22type%22%3A%22blog%22%2C%22rType%22%3A%22article%22%2C%22rId%22%3A%22140252907%22%2C%22source%22%3A%22weixin_60664036%22%7D)

CoderSuHang, 2024-07-07:
您好，我是苏航，感谢您的来信！我会尽快对来信进行查看并尽快给您进行回复，祝您身体健康，工作顺利，谢谢！