peng-zhihui/Dummy-Robot #182: 稚晖君你好，我在编译dummy电机驱动的时候报错，请问下是什么原因

haoshuiwuxiang, 2023-06-25:
====================[ Build | Ctrl-Step-STM32-fw.elf | Debug ]==================
"C:\Program Files\JetBrains\CLion 2021.2.2\bin\cmake\win\bin\cmake.exe" --build D:\STM32\Dummy-Robot-main\2.Firmware\Ctrl-Step-Driver-STM32F1-fw\cmake-build-debug --target Ctrl-Step-STM32-fw.elf -- -j 19
[  1%] Linking CXX executable Ctrl-Step-STM32-fw.elf
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Driver/tb67h450_base.cpp.obj:(.rodata._ZTV12TB67H450Base+0x1c): undefined reference to `TB67H450Base::InitGpio()'
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Driver/tb67h450_base.cpp.obj:(.rodata._ZTV12TB67H450Base+0x20): undefined reference to `TB67H450Base::InitPwm()'
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Driver/tb67h450_base.cpp.obj:(.rodata._ZTV12TB67H450Base+0x24): undefined reference to `TB67H450Base::DacOutputVoltage(unsigned short, unsigned short)'
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Driver/tb67h450_base.cpp.obj:(.rodata._ZTV12TB67H450Base+0x28): undefined reference to `TB67H450Base::SetInputA(bool, bool)'
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Driver/tb67h450_base.cpp.obj:(.rodata._ZTV12TB67H450Base+0x2c): undefined reference to `TB67H450Base::SetInputB(bool, bool)'
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Sensor/Encoder/mt6816_base.cpp.obj:(.rodata._ZTV10MT6816Base+0x14): undefined reference to `MT6816Base::SpiInit()'
d:/stm32/clion_stm32_windows/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld.exe: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/Ctrl/Sensor/Encoder/mt6816_base.cpp.obj:(.rodata._ZTV10MT6816Base+0x18): undefined reference to `MT6816Base::SpiTransmitAndRead16Bits(unsigned short)'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe[3]: *** [CMakeFiles\Ctrl-Step-STM32-fw.elf.dir\build.make:827: Ctrl-Step-STM32-fw.elf] Error 1
mingw32-make.exe[2]: *** [CMakeFiles\Makefile2:82: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/all] Error 2
mingw32-make.exe[1]: *** [CMakeFiles\Makefile2:89: CMakeFiles/Ctrl-Step-STM32-fw.elf.dir/rule] Error 2
mingw32-make.exe: *** [Makefile:123: Ctrl-Step-STM32-fw.elf] Error 2

MarPotato, 2023-07-12:
我也遇到了相同的问题，困扰我一周了，看网上说是没有include相应的头文件，我试了一下，但错误更多了，不知道老哥解决了吗？

MarPotato, 2023-07-13:
老哥，问题解决了，有小伙伴询问过智晖君，解决办法是：更换gcc-arm-none-eabi版本为9.2.1

MarPotato, 2023-07-14:
更换gcc-arm-none-eabi版本为9.2.1

MarPotato, 2023-07-14:
更换gcc-arm-none-eabi版本为9.2.1

---- 回复的原邮件 ----
| 发件人 | ***@***.***> |
| 日期 | 2023年07月14日 18:00 |
| 收件人 | ***@***.***> |
| 抄送至 | ***@***.***>***@***.***> |
| 主题 | Re: [peng-zhihui/Dummy-Robot] 稚晖君你好，我在编译dummy电机驱动的时候报错，请问下是什么原因 (Issue #182) |

重新配置运行环境

lixinswitchpi, 2023-07-17:
试试编译release模式

Tundon, 2023-08-09:
> 老哥，问题解决了，有小伙伴询问过智晖君，解决办法是：更换gcc-arm-none-eabi版本为9.2.1

或者用新版gcc-arm，手动加入那几个虚函数的空实现`virtual xxx() {}`。不过新版gcc-arm编译出来的文件大了，就没有足够空间给校验数据了。所以还是推荐换gcc-arm到9.2.1

mzrjfft, 2024-08-18:
我的烧录进去之后依旧挺大
Memory region         Used Size  Region Size  %age Used
             RAM:        7696 B        20 KB     37.58%
           FLASH:      114640 B       128 KB     87.46%
而且烧录进去之后无法点亮ref屏了

mzrjfft, 2024-08-18:
有什么可以修改flash大小吗