peng-zhihui/Dummy-Robot #184: 解决42固件最新arm gcc 编译文件过大无法存放校准数据问题

lixinswitchpi, 2023-08-14:
用最新的arm-none-eabi-gcc(12.2.1 20230214)编译42固件过大120k，会覆盖校准数据，问题出在interface_uart.cpp sscanf函数在新版中会占用50k的空间，用strtok atof替换后，固件只占79k，满足最大0x08017C00-1的要求
 FLASH:       78932 B       128 KB     60.22%

可以直接用下面代码覆盖原先的interface_uart.cpp，用clion重新编译即可

#include "common_inc.h"
#include "configurations.h"
#include <string.h>
#include <stdlib.h>

extern Motor motor;

void OnUartCmd(uint8_t* _data, uint16_t _len)
{
    float cur, pos, vel, time;
    char *token = NULL;
    switch (_data[0])
    {
        case 'c':
            token = strtok((char*) _data, "c");
            cur = atof(token);
            if (motor.controller->modeRunning != Motor::MODE_COMMAND_CURRENT)
                motor.controller->SetCtrlMode(Motor::MODE_COMMAND_CURRENT);
            motor.controller->SetCurrentSetPoint((int32_t) (cur * 1000));
            break;
        case 'v':
            token = strtok((char*) _data, "v");
            vel = atof(token);
            if (motor.controller->modeRunning != Motor::MODE_COMMAND_VELOCITY)
            {
                motor.config.motionParams.ratedVelocity = boardConfig.velocityLimit;
                motor.controller->SetCtrlMode(Motor::MODE_COMMAND_VELOCITY);
            }
            motor.controller->SetVelocitySetPoint(
                (int32_t) (vel * (float) motor.MOTOR_ONE_CIRCLE_SUBDIVIDE_STEPS));
            break;
        case 'p':
            token = strtok((char*) _data, "p");
            pos = atof(token);
            if (motor.controller->modeRunning != Motor::MODE_COMMAND_POSITION)
                motor.controller->requestMode = Motor::MODE_COMMAND_POSITION;

            motor.controller->SetPositionSetPoint(
                (int32_t) (pos * (float) motor.MOTOR_ONE_CIRCLE_SUBDIVIDE_STEPS));
            break;
        default:
            printf("Only support [c] [v] [p] commands!\r\n");
            break;
    }
}

bobzuoy, 2023-08-15:
用的 gcc-arm-none-eabi-9-2019-q4-major-win32 ，还好能编译过。

Yuz1i, 2023-09-14:
你好，我想问一下，为什么我将OnUartCmd里面的内容都注释后编译还是80%的占用呢？

lixinswitchpi, 2023-09-15:
发代码上来看看？另外，你直接用我的虚拟机环境试试 https://www.bilibili.com/video/BV1Bm4y1K7fY

Yuz1i, 2023-09-15:
> 发代码上来看看？另外，你直接用我的虚拟机环境试试 https://www.bilibili.com/video/BV1Bm4y1K7fY

好的好的，谢谢你。我的代码就是现在仓库的代码，然后将OnUartCmd函数注释。但是我是在windows环境下交叉编译的，然后我试一下你的虚拟机吧，万分感谢！！！