peng-zhihui/Dummy-Robot #188: CAN Bus 傳輸問題

Kevin75311, 2023-09-11:
稚暉君您好，
目前可以用 USB 和 REF 板子溝通，但無法透過 ref_tool CAN 跟馬達板溝通 (目前只連接 ID=1 的馬達)，
1. 有檢查過 CAN1, CAN2 initialize 都正常，馬達連接 CAN1
2. 測試連接一個馬達，終端電阻有焊接，CAN BUS 上電阻為 60 ohms
3. 無操作時 CAN_H 電壓 2.23V，CAN_L 電壓 2.20V

以下幾點想請問一下：
1. 目前接線在使用 ref_tool 時，發送指令到 joint_1 會卡住無法中止或跳出，強制下電後結果如下，請問有可能是 pyusb 版本不同造成的嗎？
[image]

2. 對於 CAN bus 的溝通，REF 是單方向傳給馬達控制板嗎？還是會有 return message 呢？

非常感謝~

lixinswitchpi, 2023-09-12:
参考一下下面的视频关于REF与Motor CAN BUS 通信例程，尝试debug motor 1 端的CAN BUS接收数据，看看能不能发现问题。另外，CAN BUS的两条线是双绞线，并行的话会出现此类问题。
https://www.bilibili.com/video/BV1E34y1K7KZ/

Kevin75311, 2023-09-13:
您好，非常感謝您，您的影片非常有幫助。
有一些電路接線的細節，想請問我理解有沒有錯誤：
1.如果接1個馬達，這樣一個 120-ohm 的電阻在電機電路板上，另外一個終端電阻要在自行焊接 (因為REF板上沒有電阻)？
2.如果接6個馬達，這樣在頭尾的電機電路板焊接120 ohm的電阻即可，其餘中間的電機電路板皆不必焊接電阻？
中間的連線接用雙絞線連接。

非常感謝您~

lixinswitchpi, 2023-09-13:
1. 是的，motor端需要接上终端电阻，稚晖君的ref上有设计终端电阻，这个需要始终enable
2. 对，CAN总线的两端都需要接120欧姆终端电阻,以匹配总线特性阻抗,保证信号质量。中间motor不需要接
我有一个改版的motor驱动板最近会发布，加上了一个switch，可以方便的打开或关闭120欧姆终端电阻，另外pcb重新布局，布线，优化了模拟信号，gnd，电源gnd等等，你可以关注我的b站

closed 2023-10-05 by Kevin75311