peng-zhihui/Dummy-Robot #191: DummyStudio的命令与源代码不一致

bobzuoy, 2023-11-25:
发现传递的参数是&，源代码中是> ,所以不联动。
这里需要修改下源代码中的文件。就可以联动了。 

https://github.com/peng-zhihui/Dummy-Robot/assets/142277489/decf0559-78f8-4b65-bd11-77090820b139

lkz-2000, 2023-12-07:
您好，请问一下这个软件能实现离线编程吗，就是输入轨迹，生成各个关节角运动

bobzuoy, 2023-12-07:
可以的。适当修CLI-Tool，应该就可以了。
可以外面放一个树莓派来实现更多功能。我改了一个python flask的网页控制。
[image]

zhbi98, 2025-05-09:
同样遇到这个问题

zhbi98, 2025-05-09:
> 发现传递的参数是&，源代码中是> ,所以不联动。 这里需要修改下源代码中的文件。就可以联动了。
> 489a2984be3a685447394f80f6845620.mp4

> 发现传递的参数是&，源代码中是> ,所以不联动。 这里需要修改下源代码中的文件。就可以联动了。
> 489a2984be3a685447394f80f6845620.mp4

没错，我同样遇到这个问题，一度怀疑是丢失了 & 字段标识的数据解析，或是在 > 标识中包含了这段解析