peng-zhihui/Dummy-Robot #198: 运动学移植的问题

Zz-er, 2024-12-10:
运动学部分似乎有一些问题，我在迁移到别的机械臂上时，就正向运动学接口来说，如下图所示首先构造Dof6Kinematic类，并且确保L_BS, D_BS, L_AM, L_FA, D_EW, L_WT的单位是m，
[image]
下面是我对正向运动学接口做的封装，确保输入的角度单位是度，转化输出的pose将xyz单位从m转换为mm，并将输出的欧拉角单位从度转换为弧度
```

Eigen::Matrix4d pzh_KinematicsSolverAdapter::fk(const std::vector<double>& joints, int* errType) 
{
	assert(joints.size() == 6);
	assert(_dhParams.size() >= 6);
	DOF6Kinematic::Joint6D_t joint_t;

	for (int i = 0; i < 6; i++)
	{
		joint_t.a[i] = (joints[i] + _dhParams[i].theta) * MathUtil::RAD2DEG;
	}

	DOF6Kinematic::Pose6D_t pose_t;

	bool flag = _kin->SolveFK(joint_t, pose_t);
	if (flag)
	{
		if (errType)
			*errType = Fk_NoError;

		Eigen::Matrix4d matrix = Eigen::Matrix4d::Identity();
		matrix.block<3, 1>(0, 3) = Eigen::Vector3d(pose_t.X*1000, pose_t.Y*1000,  pose_t.Z*1000);
		matrix.block<3, 3>(0, 0) = MathUtil::eulerAnglesToMatrix3d(Eigen::Vector3d(pose_t.A*MathUtil::DEG2RAD, pose_t.B * MathUtil::DEG2RAD, pose_t.C * MathUtil::DEG2RAD), MathUtil::EulerOrder::ZYX);
		return matrix;
	}
	else
	{
		if (errType)
			*errType = Fk_Error;

		return Eigen::Matrix4d::Identity();
	}
}
```
得到的位姿显示在场景中如图所示
[image]
我自认为已经做了比较详尽的检查了，请问我还有哪里没有检查到吗？但是无法适用，我所做的检查主要是
1. 构造函数 DOF6Kinematic(float L_BS, float D_BS, float L_AM, float L_FA, float D_EW, float L_WT)，确定参数的单位和数值
2. 正向运动学接口 bool DOF6Kinematic::SolveFK(const DOF6Kinematic::Joint6D_t& _inputJoint6D, DOF6Kinematic::Pose6D_t& _outputPose6D)，确定关节角单位为角度，转换输出位姿的单位，我已经从m转换成mm了，欧拉角也从角度转换成了我使用的弧度

============================
此外还发现代码中一个SDH表定义的构型似乎有问题，但是对后面的计算并没有影响，后面的计算中并没有访问dh表中的d和a参数
如下图所示6dof_kinematics.cpp文件中tmp_DH_matrix定义的SDH构型
[image]
将上面的构型用matlab绘制出来如下图所示
`
% 清除环境
clear all; close all; clc;

% 注意：Link的参数顺序为 [theta d a alpha]
L1 = Link([0        0.550  0.180  -pi/2]);
L2 = Link([-pi/2    0    0.680   0   ]);
L3 = Link([pi/2        0.165    0  pi/2]);
% L3 = Link([pi/2        0    -0.165  pi/2]); % L3的DH参数应该定义成这样
L4 = Link([0         0.842  0    -pi/2]);
L5 = Link([0         0    0    pi/2]);
L6 = Link([0         0.12    0    0   ]);
% 创建机器人模型，将所有链节按顺序组合
SDH_robot = SerialLink([L1 L2 L3 L4 L5 L6], 'name', 'SDHRobot');
view(3) 

% SDH_robot.plot([0 0 0 0 0 0]);

% 设置机器人绘图参数（可选）
SDH_robot.display

SDH_robot.teach([0 -pi/2 pi/2 0 0 0]);`
[image]
D_ELBOW参数和示意图中描述的不同
[image]
启用上面matlab代码中的注释，将L3定义为 L3 = Link([pi/2        0    -0.165  pi/2]);
[image]
虽然对后面的计算没有影响，应该定义为
```
    float tmp_DH_matrix[6][4] = {
        // theta            d                   a                   alpha
        {0.0f,            armConfig.L_BASE,    armConfig.D_BASE, -(float)M_PI_2},
        {-(float)M_PI_2, 0.0f,                armConfig.L_ARM,  0.0f},
        {(float)M_PI_2,  0.0f,                 -armConfig.D_ELBOW, (float)M_PI_2},
        {0.0f,            armConfig.L_FOREARM, 0.0f,             -(float)M_PI_2},
        {0.0f,            0.0f,                0.0f,             (float)M_PI_2},
        {0.0f,            armConfig.L_WRIST, 0.0f, 0.0f}
    };
```
3.

linqingjue, 2025-07-24:
3关节图示状态是90°状态，并且根据上位机来看规定的正向旋转判断Z轴，我得出来的旋转矩阵`float tmp_DH_matrix[6][4] = {
        {0.0f,            armConfig.L_BASE,    armConfig.D_BASE, -(float) M_PI_2},
        {-(float) M_PI_2, 0.0f,                armConfig.L_ARM,  0.0f},
        {0,               0.0f,                armConfig.D_ELBOW, (float) M_PI_2},
        {0.0f,            armConfig.L_FOREARM, 0.0f,             (float) M_PI_2},
        {0.0f,            0.0f,                0.0f,             -(float) M_PI_2},
        {0.0f,            armConfig.L_WRIST,   0.0f,             0.0f}
    };`

linqingjue, 2025-07-24:
例图里关节3是90°状态，我根据上位机角度旋转方向判断的Z轴朝向，最后得到的矩阵是这个：
```
float tmp_DH_matrix[6][4] = {
        {0.0f,            armConfig.L_BASE,    armConfig.D_BASE, -(float) M_PI_2},
        {-(float) M_PI_2, 0.0f,                armConfig.L_ARM,  0.0f},
        {0,               0.0f,                armConfig.D_ELBOW, (float) M_PI_2},
        {0.0f,            armConfig.L_FOREARM, 0.0f,             (float) M_PI_2},
        {0.0f,            0.0f,                0.0f,             -(float) M_PI_2},
        {0.0f,            armConfig.L_WRIST,   0.0f,             0.0f}
    };
```